CIDCO-dev/Poseidon #248: [sonar_nmea-0183_tcp_client] Add use* parameters

glabmoris, 2021-11-17:
Add the following (boolean) parameters to both the DeviceNmeaClient and NetworkNmeaClient constructors:

useSonar
useGnss
useAttitude

By default, only useSonar should be true, the others  (useGnss, useAttitude) should be false.

patcmorneau, 2021-11-17:
je suis pas sur de bien comprendre, il me semble que c'est déjà comme ça mais avec des nom de variable différent
```

		bool useDepth = true;
		bool usePosition = false;
		bool useAttitude = false;
```

pour faire mes tests j'utilisais le stack : hydrobox_nmeadevice.lauch et j'enlevais les nodes rasberry_vitals et gnss_mosaic.
Si je veux tester ma condition de logging, quelle node suis-je le mieux d'utiliser ?

glabmoris, 2021-11-17:
Le nmea_device_node est le mieux. Par paramètres, on parle de paramètres ROS (via getParam). Les paramètres extraits avec getParam vont venir populer les variables:

bool useXXX = ros::getParam(....);

patcmorneau, 2021-11-17:
comment que je fais pour populer : bootstrappedGnssTime
c'est la variable qui me manque pour confirmer ou non que ca fonctionne
```
bool Writer::toggleLogging(logger_service::ToggleLogging::Request & request,logger_service::ToggleLogging::Response & response){
	mtx.lock();

	if(bootstrappedGnssTime){
		std::cerr<<" Gnss ok \n";
		if(!loggerEnabled && request.loggingEnabled){
			//Enabling logging, init logfiles
			loggerEnabled=true;
			init();
		} 
```
les parametres sont passés comme il faut

```
PARAMETERS
 * /Sonar/usePosition: True
 * /logger_text/speed_threshold: 1.0
```

glabmoris, 2021-11-17:
It's done automatically through gnssCallback defined in src/logger_text.cpp (line 64):

```
void Writer::gnssCallback(const sensor_msgs::NavSatFix& gnss){

	if(!bootstrappedGnssTime && gnss.status.status >= 0){
		bootstrappedGnssTime = true;
	}

	if(bootstrappedGnssTime && loggerEnabled){
        (...)
```

It's registered in the main in nodes/logger_text_node.cpp (line 22):

```
ros::Subscriber sub1 = n.subscribe("fix", 1000, &Writer::gnssCallback,&writer);
```

glabmoris, 2021-11-22:
Clo

closed 2021-11-22 by glabmoris