CIDCO-dev/Poseidon #261: Logging Modes

glabmoris, 2022-01-14:
Support 3 loggin modes:

- Manual: We toggle logging on/off
- Speed-based : We trigger logging on/off with speed
- Always On:  we are always on

Ddoiron-cidco, 2022-01-14:
`<div class="dropdown no-arrow mb-4">
<button class="btn btn-secondary dropdown-toggle" type="button" id="dropdownMenuButton" data-toggle="dropdown" aria-haspopup="true" aria-expanded="false"> Dropdown (no arrow) </button>
	<div class="dropdown-menu" aria-labelledby="dropdownMenuButton">
		<a class="dropdown-item" href="#">Action</a>
        <a class="dropdown-item" href="#">Another action</a>
        <a class="dropdown-item" href="#">Something else here</a>
    </div>
</div>`

glabmoris, 2022-01-17:
> ```
> <div class="dropdown no-arrow mb-4">
> <button class="btn btn-secondary dropdown-toggle" type="button" id="dropdownMenuButton" data-toggle="dropdown" aria-haspopup="true" aria-expanded="false"> [...]

The problem with the dropdown button widget is that it doesn't disply the currently selected item. It's good to pack a lot of action links into a single dropdown, but doesn't do wso as efficiently as a good old <select>

patcmorneau, 2022-01-17:
je ne sais pas si c'est toujours a jour ca mais maintenant le code
ressemble plus a ca et le mode sélectioné vient du config file...

> function processConfig(config){
> var form = document.getElementById('formContent');
>
> form.innerHTML='';
> var newHTML='';
> var endNewHTML='';
>
> config.forEach(function (item,index){
> if(item.key == "loggingMode"){
> endNewHTML += '<div class="form-group row"><label class="col-lg-2
> col-form-label" for="' + item.key +'">'+ item.key + '</label><div
> class="col-lg-10"><select id="'+ item.key  + '" name="logginModeSelector"
> class="dropdownright configurationField"></div>';
>
> //add validation ?
> var mode = Number(item.value);
> switch(mode){
> case 1:
> endNewHTML += '<option value="1" selected>always</option><option
> value="3">manual</option><option value="3">speed
> based</option></select></div></div>';
> break;
> case 2:
> endNewHTML += '<option value="1">always</option><option value="2"
> selected>manual</option><option value="3">speed
> based</option></select></div></div>';
> break;
> case 3:
> endNewHTML += '<option value="1">always</option><option
> value="2">manual</option><option value="3" selected>speed
> based</option></select></div></div>';
> break;
> default:
> console.log("erreur")
> //set an alert or something
> endNewHTML += '<option value="1" selected>always ON</option><option
> value="2">manual</option><option value="3">speed
> based</option></select></div></div>';
> break;
> }
> }
> else{
> newHTML += '<div class="form-group row"> <label class="col-lg-2
> col-form-label" for="' + item.key +'">' + item.key + '</label> <div
> class="col-lg-10"> <input id="'+ item.key  + '" class="form-control
> text-right configurationField" type="text" value="' + item.value + '"/>
> </div></div>';
> }
>
> });
> newHTML += endNewHTML;
> form.innerHTML = newHTML;
> }
>
Par Contre il y a un problème intermittent ... le fichier config.txt n'est
pas lu correctement ou le problème est ailleurs(race condition) mais des
fois quand je lance roslaunch ca me fait ca (pas souvent):

> [ INFO] [1642434241.172966196]: getting speed threshold configuration...
> [ WARN] [1642434241.174192383]: no speed threshold define in config file,
> defaulting to 5 Kmh
>

patcmorneau, 2022-03-09:
can be close

Ddoiron-cidco, 2022-05-27:
@glabmoris  to be close

glabmoris, 2024-10-21:
Fixed.

closed 2024-10-21 by glabmoris