CIDCO-dev/Poseidon #264: Améliorer robustesse mémoire flash

tomypineau, 2022-01-26:
En général les carte SD sont designées pour rouler un OS (plusieurs écritures dans le même secteur) alors que les clés USB sont utilisées pour du stockage (plusieurs écritures dans des secteurs différents). Il est donc possible d'obtenir des cartes SD spécialisées qui permettent beaucoup plus d'écriture dans le même secteur (60000 vs 3000 pour la flash) avant corruption. Cela n'existe pas au niveau USB. Les USB "industriel" vont souvent garantir la rapididé, mais ne seront pas beaucoup supérieur en terme de corruption.

Il pourrait donc être intéressant de bouger la section OS vers une carte SD et si c'est possible, mettre cette partition en lecture seul. Le data quant à lui pourrait être sur le USB sans trop de problèmes. 

https://www.ni.com/en-ca/support/documentation/supplemental/12/understanding-life-expectancy-of-flash-storage.html
https://www.embeddedts.com/assets/preventing-filesystem-corruption-in-embedded-linux

glabmoris, 2022-01-27:
Merci Tomy,

Ce sont des points pertinents. Je déclinerais ce dernier en les questions suivantes:

 - Est-on en mesure de quantifier la différence de vulnérabilité à la corruption des deux solutions? Le systeme peut facilement supporter l'une ou l'autre des options.
 
 - As-tu des suggestions de cartes SD qui seraient plus robustes que la SwissBit proposée (https://www.digikey.ca/en/products/detail/swissbit/SFU3064GC2AE2TO-I-LF-1A1-STD/12181130)? Un de ses features clé est la feature hotplug sans corruption, ce qui n'est pas le cas des cartes SD qui ont une fenêtre d'opportunité de corruption par design lors d'une coupure de courant au milieu d'un cycle de lecture/écriture. Car au delà des comparatifs sur l'endurance de la mémoire NAND utilisée, c'est surtout le fait qu'on ne peut pas contrôler quand l'alimentation sera coupée qui pose le risque de corruption le plus grand si on considère que l'appareil peut être utilisé dans tous genres d'embarcations (...à mon avis...ça serait intéressant d'en discuter).
 
 - L'option d'une partition en lecture seule est définitivement une amélioration intéressante. Cependant, ceci exige le maintien d'une configuration d'Ubuntu sur mesure, ainsi que de doubler les coûts des appareils de stockage.  Je crée un issue à cet effet, le #268

glabmoris, 2024-10-21:
Fixed

closed 2024-10-21 by glabmoris