CIDCO-dev/Poseidon #265: Ajout d'un RTC

tomypineau, 2022-01-26:
Pour que le système puisse avoir une base de temps si le signal GPS n'est pas fonctionnel.

Nous avons déjà utilisé un DS3231MZ avec un pi qui fonctionnait bien.

glabmoris, 2022-01-28:
C'est un ajout intéressant, il faudrait ajouter de la logique autour de chrony pour inclure la synchro de la RTC au temps satellite (via rtcsync) et l'inverse (RTC vers sysclock) lorsqu'on n'a pas de fix GPS/PPS et que l'horloge système n'est pas à l'heure (car si on fait juste perdre le fix GPS, l'horloge systeme continue de fonctionner, seulement avec des risques de drift).

Ddoiron-cidco, 2022-02-23:
Test with GPS stream cut

before disconnecting the antenna
[image]

during the antenna disconected
[image]

when reconnecting the antenna
[image]

the gps is in fix 3D
[image]

after 1 min
[image]

after 5 min
[image]

glabmoris, 2022-02-23:
Done in #278

closed 2022-02-23 by glabmoris