CIDCO-dev/Poseidon #294: [GNSS-ZED-F9P node] error in ubx files generation

Ddoiron-cidco, 2022-05-18:
Acording to the configuration of the gnss the rawx and sfrbx are transmited on the usb port
[image]

The ubx file cant be converted by rtkconv 
[image]

glabmoris, 2022-05-18:
That's a great find, it's clearly byte-ordering error, probably from developing on x86_64 (little-endian) to run on ARM (big-endian)

patcmorneau, 2022-05-18:
Thanks for the great documentation on the bug, working on it ASAP.

patcmorneau, 2022-05-19:
ubx file are now readablewith this fix:
L.211 -> gnss_zed_f9p.h
```
//write frame to bin file
uint8_t sync[2];
sync[0] = 0xb5;
sync[1] = 0x62;
```

but we are still missing observation packets ( b5 62 02 15 )

patcmorneau, 2022-05-19:
when the payload is bigger than 255 bytes, there's a byte ordering problem and the packets doesn't get written.
the payloads always exceed 255 bytes for observations packets

glabmoris, 2022-05-23:
Seems like a bug combo. lscpu shows that it's running in little-endian, so it's not an endianness bug, it appears that the bytes were just badly swapped at write time and that @patcmorneau 's fix solved that piece of the puzzle. 

Furthermore, there was an issue in the binary read where the payload was not properly read. I noticed that we are using the read() syscall, which may not always return the proper amount of bytes. I've added a readSerial() method to handle this problem. Adding a check on incoming packets to find observation packets (b5 62 02 15) to print them out gives the expected result: Houston, we have obervation packets.

I'm going to be on the road, so could you please test that everything works with rtkconv and commit the codebase on the test hydroblock once confirmed?

Good job everyone!

glabmoris, 2022-05-24:
Great!

Can we generate a RINEX through rtkconv?

Ddoiron-cidco, 2022-05-24:
the obs file was send to NRCAN for PPP

i'am waiting for the result

Ddoiron-cidco, 2022-05-24:
i just recive the repport
[2022.pdf](https://github.com/CIDCO-dev/Poseidon/files/8765797/2022.pdf)

Ddoiron-cidco, 2022-05-27:
fixed

closed 2022-05-27 by Ddoiron-cidco