CIDCO-dev/Poseidon #345: Service executed by root

Ddoiron-cidco, 2023-08-31:
Services are executed by root should be executed by a user

Recommendation :

Clone github on current user
Create user Poseidon and copy workspace and www under poseidon home
Chown file and directory to the user Poseidon.
Copy the configuration file and calibration in the Home of poseidon if the are not present
Modify the service file for lighttpd to be launch by the user

patcmorneau, 2024-09-12:
in order the user to be able to communicate with i2c devices
```
sudo nano /etc/udev/rules.d/99-i2c.rules
KERNEL=="i2c-1", SUBSYSTEM=="i2c-dev", MODE="0660", OWNER="Poseidon"
sudo udevadm control --reload-rules
sudo udevadm trigger
```

glabmoris, 2024-10-21:
We need to have root access for GPIO access with the current ROS version.

closed 2024-10-21 by glabmoris