CIDCO-dev/Poseidon #359: [Diagnostics] Ajouter test sur les horloges

glabmoris, 2023-10-27:
Ajouter un test de diagnostic pour la condition suivante:
On veut vérifier que le temps GNSS est le même que le temps système (+- 1ms)

Ddoiron-cidco, 2023-10-31:
Jai fait un petit script de test al nuit passer et je pense que ca pourais etre utile. 

c'est un script python 
on as a installer gpsd-py3 avec pip

```
import gps
import time
from datetime import datetime

# Connectez-vous à GPSD
session = gps.gps("localhost", "2947")
session.stream(gps.WATCH_ENABLE | gps.WATCH_NEWSTYLE)

try:
    while True:
        report = session.next()
        if report['class'] == 'TPV':
            if hasattr(report, 'time'):
                gps_time = report.time
                system_time = time.time()
                time_difference = gps_time - system_time
                gps_time_str = datetime.utcfromtimestamp(gps_time).strftime('%Y-%m-%d %H:%M:%S')
                system_time_str = datetime.utcfromtimestamp(system_time).strftime('%Y-%m-%d %H:%M:%S')
                print(f"GPS Time: {gps_time_str}")
                print(f"System Time: {system_time_str}")
                print(f"Time Difference (seconds): {time_difference}")
                break

except KeyboardInterrupt:
    session.close()

```

patcmorneau, 2023-11-02:
`chronyc sources`  nmea donne +/- 101 ms...

Et le test dans le PR#305 donne en moyenne 500 ms

Ddoiron-cidco, 2024-05-07:
@glabmoris 

https://github.com/Ddoiron-cidco/Poseidon/blob/0b595d9a41a7ae311224cf4acdaf5372db666673/install/stages/5-finalize.sh#L26C1-L26C90

offset 0.5 seconde = 500ms

the delay could be adjusted too.

glabmoris, 2024-10-21:
Set this test to < 30ms