CIDCO-dev/Poseidon #369: Conversion de secondes en micro secondes: (double vers uint64_t)

MHS1985, 2023-12-14:
In line 21 of georefLog.cpp we can notice a conversion of double to uint64_t                        

             hdrPosition.packetTimestamp = static_cast<uint64_t>(entry->time * 1000000);

This conversion can result the lost of accurate timestamp data !

patcmorneau, 2024-01-24:
I would like to have more information on how to solve the possible lost of accuracy.

so i made a basic c++ code to verify your concerns:
```
double x = 12345678910.0123456789;
	std::cout<< x <<"\n";
	
	double y = x*1000000;
	std::cout<<y<<"\n";
	
	auto z = static_cast<uint64_t>(y);
	std::cout<<z<<"\n";
	
	auto k = static_cast<uint64_t>(x*1000000);
	std::cout<<k<<"\n";
```
the output is :
```
1.23457e+10
1.23457e+16
12345678910012346
12345678910012346
```

it performs as expected...

the documentation for the entry->time can be found on line 24 in this page : https://github.com/CIDCO-dev/SBET-decoder/blob/master/src/SbetProcessor.hpp

the biggest number in entry time will be 604800, as the documentation says its seconds... no smaller units should be stored
unless its not documented properly.

glabmoris, 2024-10-21:
Close due to inactivity.

closed 2024-10-21 by glabmoris