CIDCO-dev/Poseidon #370: [Sonar nmea] serial port speed

Ddoiron-cidco, 2024-01-18:
Add an argument on the config file and the web page to set the serial port speed

patcmorneau, 2024-01-25:
we could set the baud rate this way before opening it
```
#include <stdio.h>
#include <fcntl.h>
#include <termios.h>
#include <unistd.h>

int main() {
    int serial_fd = open("/dev/ttyS0", O_RDWR | O_NOCTTY);
    if (serial_fd == -1) {
        perror("Error opening serial port");
        return 1;
    }

    struct termios tty;
    if (tcgetattr(serial_fd, &tty) != 0) {
        perror("Error getting attributes");
        return 1;
    }

    cfsetospeed(&tty, B9600);  // Set output baud rate to 9600
    cfsetispeed(&tty, B9600);  // Set input baud rate to 9600

    if (tcsetattr(serial_fd, TCSANOW, &tty) != 0) {
        perror("Error setting attributes");
        return 1;
    }

    close(serial_fd);
    return 0;
}
```

glabmoris, 2024-10-21:
Fixed

closed 2024-10-21 by glabmoris