CIDCO-dev/Poseidon #382: Geofencing

glabmoris, 2024-05-09:
Add a geofence feature to enforce specific zones to acquire data to avoid collecting illegal data
 - Ideally done at the HB level, not the portal

jomad-gst, 2024-07-16:
I have a preliminary Geofence implemented in https://github.com/jomad-gst/Poseidon/tree/geofence

jomad-gst, 2024-07-17:
Some questions that needs to be answered at some point:

1. How does the Geofence interact with existing logging methods? The speed logging currently always attempt to start logging when the speed is fast enough, this could interfere with a geofence and they would be stuck in a loop turning on and off.  What happens if the logging mode is manual and the logging is manually turned on while outside the Geofence? Maybe the simplest solution is to have another logging mode. Atleast the interaction should be well defined.
3. Do we need some dependency to parse wkt and test for "point in polygon"? The current implementation provides a naive simple implementation. The naive approach is fine for most usecases I think.

Jonasmadsen, 2024-08-13:
Makes sense , I think I have a working version now, on the branch https://github.com/jomad-gst/Poseidon/tree/geofence if you create a feature branch for it I will make a pull request to it, or you can just copy my code into a new branch in your repository.

Feel free to edit the code and use just the parts that you find useful.

closed 2025-01-20 by glabmoris